duckietown/docs-opmanual_sky #135: 7.1 Shows different cable

coachsimard, 2020-07-23:
In Unit A-7, 7.1, Prep IR Sensor Wire and ADC, picture shows and asks to cut the black connector. Wiring harness supplied does not have that and is in fact already tinned on the stripped end.

GarrettWarren, 2020-08-31:
Could you explain a bit more? Did you IR sensor look like this one: https://docs.duckietown.org/daffy/opmanual_sky/out/build_materials_included.html#sub:materials-ir ?

coachsimard, 2020-08-31:
Yes, mine did.

GarrettWarren, 2020-08-31:
great! what did you mean by "wiring harness supplied does not have that". i know its been a while since the build. Just want to make sure that I understand this issue so that I can modify the docs accordingly. thanks for helping us improve them!

coachsimard, 2020-08-31:
I had an issue with the first one, and Peter gave me the old version. The
first one was different.

*Sincerely,*

*Rick Simard**Technology Faculty*

*Mount Saint Charles Academy**800 Logee St*
*Woonsocket, RI 02895**-5599*
*Office: 401-769-0310*
*Fax: 401-762-2327*